powsybl/powsybl-open-rao #548: UCTE GLSK importer

annetill, 2021-11-18:
* **Do you want to request a *feature* or report a *bug*?**

Just a feature ! 

* **What is the current behavior?**

The UCTE GLSK importer is inside `farao-core`.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem**

* **What is the expected behavior?**

I ask you formally the right to move the code from farao-core to powsybl, under the same licence and keeping the original authors. Just tell me !

* **What is the motivation / use case for changing the behavior?**

* **Please tell us about your environment:**
  - FARAO Version: ...
  - OS Version: ...

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, spectrum, etc)

(if a question doesn't apply, you can delete it)

obrix, 2021-11-19:
After some discussions, we went for a complete move of the glsk modules to powsybl-entsoe. Not juste ucte related. The only exception is the glsk-virtual-hubs module which has some specific dependencies for which a move would not be relevant.

obrix, 2021-11-22:
I just added a branch on my farao-core fork to illustrate the impact of the glsk related code move to powsybl-entseo. See [here](https://github.com/obrix/farao-core/tree/glsk_moved_to_powsybl).

closed 2023-04-24 by pet-mit