powsybl/powsybl-open-rao #908: Use TwoSides and ThreeSides

annetill, 2024-03-01:
### Describe the current behavior

Before using the `powsybl-core` remedial actions in powsybl-open-rao, we have to think of a small refactoring to remove Side.LEFT and Side.RIGHT to TwoSides enum (an maybe ThreeSides).  

### Describe the expected behavior

- The question of CRAC JSON breaking change in its format must be decided. 
- If yes, the question of backward compatibility has to be decided. 
- Documentation must be adapted (all website examples).

pjeanmarie, 2024-05-31:
Do we update Crac json ("side" : "left")?
also parameter json ("default-monitored-line-side" : "monitor-lines-on-left-side")?
also RaoResults json ("leftSide" : {...)?

pjeanmarie, 2024-05-31:
com.powsybl.openrao.data.cracapi.cnec.Side no longer exist, is it a breaking change?

phiedw, 2024-06-05:
I would say yes replace also in the json files, while keeping retro compatibility.
And it probably is indeed a breaking change for some parts of gridcapa.

pjeanmarie, 2024-06-07:
Updated json crac 2.4 and new version 1.5 for json raoResult.
But for the json crac creation parameters with "default-monitored-line-side" with values "monitor-lines-on-left/right/both-side", it seems this file do not have version system, can I keep it like that or do I have to add a version system fo this files two?

PR #1027 (2024-06-27): Use twosides and threesides
**Please check if the PR fulfills these requirements**

- [ ] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**

Fixes #908

**What kind of change does this PR introduce?**

feature

**What is the current behavior?**

Rao has its enum Side with values LEFT/RIGHT and a conversion method to take equivalent core (iidm) TwoSides ONE/TWO

**What is the new behavior (if this is a feat [... cut]
closed 2024-06-27 by pet-mit