powsybl/powsybl-open-rao #1020: Make relative ranges more precise in MIP

phiedw, 2024-05-23:
### Describe the current behavior

Currently relative tap ranges in MIP are modeled via their setpoint, limiting the distance between the setpoint to the smallest tap step on the pst multiplied by the number of tap availables.

### Describe the expected behavior

We should create constraints between the tap variation variables (and the initial tap) instead if we are not using continuous approximation.

Maybe we should probably also use the tap variations when reading the solution of the problem instead of the setpoint and then rounding it, if not using continuous approximation.

### Describe the motivation

The current behaviour is not very precise, because the taps are usually larger when close to 0. So for instance if the smallest step is of 1° (close to the end of the pst), but the steps close to tap 0 are around 5°, and if we're initially at tap 0 with a relative range of +-10 taps, then we'll only be able to move the tap by 1 or 2 taps instead of 10.

phiedw, 2024-05-24:
Even while working with Jeremy for the implementation of relative_to_previous_timestep ranges in the MIP, on a very basic case, we found that implementing the range with the tap gave better results than the setpoint approximation.

bqth29, 2024-05-31:
I implemented such variation variables in #1009 so I think some code could be reused for the purpose of this issue

closed 2024-07-04 by pet-mit