powsybl/powsybl-open-rao #1046: RaoResult: allow exporting flow values in %Imax unit

pet-mit, 2024-06-20:
### Describe the current behavior

When exporting a RaoResult object, only AMPERE and MEGAWATT units are supported for flow values

### Describe the expected behavior

Also allow wrting flows in %Imax, making all PhysicalParameter.FLOW units supported

### Describe the motivation

Genericity + using %Imax makes it easier to see important overloads on the network

phiedw, 2024-06-21:
Do you think some processes could also be interested in a %Imax objective function?

pet-mit, 2024-06-21:
> Do you think some processes could also be interested in a %Imax objective function?

maybe, but it would be a very big effort to implement such an objective function IMO

pet-mit, 2026-11-16:
closed as not really needed ; #2003 should cover minimizing % overloads

closed 2026-11-16 by pet-mit