powsybl/powsybl-open-rao #1054: Cost minimization mode

phiedw, 2024-06-21:
### Describe the current behavior

Currently the objective function in the RAO is the maximization of the minimum margin.

### Describe the expected behavior

Once #1053 is done, it will be interesting to be able to minimize the cost, while making sure the network is secure.
(new objective function would only include the cost of the actions, and the minimum margin would be a new constraint for the mip to make sure it's positive)

### Describe the motivation

Would be used in some processes to replace the dichotomy (for performance issues)

bqth29, 2024-10-25:
I suggest we split this issue in 2 PRs: one for range actions and one for network actions

bqth29, 2024-10-25:
Here is a summary of the work to do for this issue (given that the proposed cost API from #1182 is accepted). We will consider that only **range actions** have a cost for the moment. Network actions require a change in the evaluation of leaves in the search-tree and will have their dedicated thread.

# Range Actions costs

Based on the work carried out in #1182, range actions now have to types of costs:
1. an **activation cost** which is absolute and correspond to the money to spent to activate the remedial action, no matter how much the range is changed
2. **variation costs** (one for an upward variation and one for a downward variation) that correspond to the money required to change the set-point of the remedial action per range unit (ex: MW/€); the total variation cost depends on how much the range was changed both in direction (upward/downward) and absolute value

> If the same range action is activated several times, say in preventive and in curative or during several curative instants, should it be counted as activated only once or several? If several, how many?

For a given range action $r$, we will denote the activation cost as $c_{act}(r)$ and the upward and downward variation costs respectively as $c_{+}(r)$ and $c_{-}(r)$.

# State cost

Each optimization state $s$ must be associated to a cost that we will denote as $C(s)$ that corresponds to the money that must be spent to move the network from $s$'s previous state to $s$.

bqth29, 2024-10-25:
Other observation: if a range action has no variation cost defined on at least one of the two variation directions, we should use the penalty costs defined the RAO parameters to keep a similar behaviour as in min margin optimization. That way, set-point are not changed randomly by the RAO because of null costs.

closed 2024-12-10 by phiedw