powsybl/powsybl-open-rao #1081: Crac import: move offsetDateTime to CracCreationParameters

pet-mit, 2024-07-05:
### Describe the current behavior

Almost all Crac creation parameters are in the CracCreationParameters object, except offsetDateTime (probably because it was the first parameter created). Moreover, this parameter is only useful for some native Crac formats (not useful for json, nor for cse).

### Describe the expected behavior

It could be a good idea to move it to CracCreationParameters' extensions where needed.

### Describe the motivation

Clearer API

bqth29, 2025-01-23:
What about we also include the timestamp directly in the CRAC API (as it done for networks). That could simplify further works on inter-temporality.

pet-mit, 2025-01-24:
> What about we also include the timestamp directly in the CRAC API (as it done for networks)? That could simplify further works on inter-temporality.

Currently it exists in the CracCreationContext. If it's not enough, I personnally am not against adding it to CRAC

bqth29, 2025-02-06:
For JSON import, we'll have to decide which TS prevails: the one from the JSON data or the one from the CRAC creation parameters

Godelaine, 2025-02-06:
will resolve problematic from PR1265 to prevent confusion with intertemporal timestamps
Gridcapa : post PR, create a test to show how to import a cim crac without timestamp + with updated crac parameters

PR #1274 (2025-02-14): Add offsetdatetime in cracparameter
**Please check if the PR fulfills these requirements**

- [ ] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**

Fixes #1081 

**What kind of change does this PR introduce?**

**Does this PR introduce a new Powsybl Action implying to be implemented in simulators or pypowsybl?**
- [ ] Yes, the corresponding issue is [here](link)
- [ ] No

**What is the current behavior?** [... cut]
closed 2025-02-14 by bqth29