powsybl/powsybl-open-rao #1088: Take in account RA Usage Limits in 2P Global Opt

bqth29, 2024-07-10:
### Describe the current behavior

When running a second preventive RAO, RA usage limits are not considered (at least for curative instants).

### Describe the expected behavior

Take in account all RA usage limits in 2P GO.

bqth29, 2024-07-11:
After some investigations, I think that this issue can be solved by adding some additional variables and constraints to the MIP if we are running the RAO in 2P Global Opt mode. The preventive RA usage limits can be kept for the 2P search tree and preventive range actions.

However, the curative RA usage limits must be passed as MIP data because all curative range actions are optimized at the same time. Below is a summary of the changes one could make to take in account curative RA usage limits.

> **Note:** some mathematical constants are defined in the [dedicated RA usage limits MIP page](https://powsybl.readthedocs.io/projects/openrao/en/stable/castor/linear-problem/ra-usage-limits-filler.html) in our online documentation.

In the following we will only consider curative states.

Finally, as a reminder, RA usage limits are **cumulative** among instants of the same kind (e.g. among curative instants).

# Preliminary notations

Let $s$ be a curative state and $tso$ be a TSO.

- $\tau(s)$ is the number of topological remedial actions that were applied at state $s$ during the first RAO
- $\tau_{TSO}(s, tso)$ is the number of topological remedial actions of TSO $tso$ that were applied at state $s$ during the first RAO
- $e(s, tso)$ is the number of elementary actions of TSO $tso$ that were applied at state $s$ during the first RAO

Besides, for a given curative state $s$, we will denote $\mathcal{S}_{prev}(s)$ the set of states that:

- are curative
- share the same contingency as $s$
- occur before $s$ (i.e. their associated instant $i'$ comes before $i$)

> Note that $s \in \mathcal{S}^{prev}(s)$

Finally, $t(s)$ denotes the number of TSOs that have applied a remedial action at state $s$ or in any previous curative state (for the same contingency).

# Maximum number of remedial actions

Let $i$ be a curative instant The maximum number of remedial actions that can be used at the instant $i$ is $RA_{\max}(i)$ and is defined in the CRAC.

In 2P Global Opt, we are optimizing all curative states at the same time which means that states associated to different contingencies are being considered in the same MIP model so the $RA_{\max}(i)$ has to be duplicated for each contingency scenario to keep the independence among these contingency scenarios.

Thus, for each state $s$ and its associated curative instant $i$, we define the maximum number of remedial actions applicable for this state as $RA_{\max}^{*}(s) = RA_{\max}(i) - \tau(s)$.

To account for the cumulative behavior of this RA usage limit among the curative instants, we need to add the following constraints to the MIP: 

$$\forall s \text{ curative}, \sum_{s' \in \mathcal{S}^{prev}(s)} \sum_{r \in \mathcal{RA}} \delta(r, s') \leq RA_{\max}^{*}(s)$$

# Maximum number of remedial actions per TSO

Similarly to the previous section, we introduce for all state $s$ (with associated instant $i$) and TSO $tso$ the maximum number of remedial actions applicable at state $s$ for TSO $tso$: $RA_{\max}^{TSO *}(s, tso) = RA_{\max}^{TSO}(i, tso) - \tau_{TSO}(s, tso)$; and we add the following constraints:

$$\forall s, \forall tso, \sum_{s' \in \mathcal{S}^{prev}(s)} \sum_{r \in \mathcal{RA}(tso)} \delta(r, s') \leq RA_{\max}^{TSO *}(s, tso)$$

# Maximum number of PST  per TSO

We introduce for all state $s$ (with associated instant $i$) and TSO $tso$ the maximum number of PST remedial actions applicable at state $s$ for TSO $tso$: $PST_{\max}^{TSO *}(s, tso) = PST_{\max}^{TSO}(i, tso)$; and we add the following constraints:

$$\forall s, \forall tso, \sum_{s' \in \mathcal{S}^{prev}(s)} \sum_{r \in \mathcal{RA}(tso), r \text{ is a PST}} \delta(r, s') \leq PST_{\max}^{TSO *}(s, tso)$$

# Maximum number of elementary actions per TSO

We introduce for all state $s$ (with associated instant $i$) and TSO $tso$ the maximum number of elementary actions applicable at state $s$ for TSO $tso$: $EA_{\max}^{TSO *}(s, tso) = EA_{\max}^{TSO}(i, tso) - e(s, tso)$; and we add the following constraints:

$$\forall s, \forall tso, \sum_{s' \in \mathcal{S}^{prev}(s)} \sum_{r \in \mathcal{RA}(tso), r \text{ is a PST}} \Delta t(r, s') \leq EA_{\max}^{TSO *}(s, tso)$$

# Maximum number of TSOs

We introduce for all state $s$ (with associated instant $i$) the maximum number of TSOs that can apply a remedial action at state $s$ or before: $TSO_{\max}^{*}(s) = TSO_{\max}(i) - t(s)$.

We already have the variables that indicate if a TSO $tso$ has a range action activated at state $s$, denoted $\delta^{TSO}(tso, s)$. We now introduce a _cumulative_ binary variable $\delta^{TSO +}(tso, s)$ that indicates whether $tso$ activated a remedial action of its own during $s$ or any previous state (for the same contingency). This implies to add the constraints:

$$\forall s, \forall tso, \delta^{TSO}(tso, s) \leq \delta^{TSO +}(tso, s)$$

And we can finally add the RA usage limit constraints:

$$\forall s, \sum_{s' \in \mathcal{S}^{prev}(s)} \sum_{tso} \delta^{TSO +}(tso, s') \leq TSO_{\max}^{*}(s)$$

Godelaine, 2024-11-29:
update : still needed for all listed RA Usage limits

Godelaine, 2026-01-19:
plan de dev :
1) introduire le passage instant -> state, ce qui conduira au fix du bug potentiel actuel (rappel: surestimation de parades curatives dans le 2P). Faire 1 test avec overshoot de curatif actuel, le voir fixed avec cette PR. 2 semaines

2) introduction des équations pour le multi curatif et l'effet cumulatif. beaucoup de tests cucumber à faire : en 2P. vérifier qu'on est couvert par les tests classiques hors 2P. pour csa. 3 semaines.