powsybl/powsybl-open-rao #1104: RaoParameters.read()/.write()

bqth29, 2024-08-01:
### Describe the expected behavior

To follow the work carried out for CRAC and RaoResult import and export, the RAO parameters should be imported / exported using read / write methods

pet-mit, 2024-08-06:
**_do not start working on this before completion of https://github.com/powsybl/powsybl-open-rao/pull/1101, in order to avoid conflicts_**

bqth29, 2025-02-03:
Currently, we have to RAO parameters files format:
- JSON
- YAML

The question is: what do we want to do with YAML configs? Do we keep using the `RaoParameters.load()` and only use `RaoParameters.read` for JSON parameters?

Otherwise, shall we crate a `YamlRaoParametersImport` class that is an `AutoService` of `Importer` and replace `load` by `YamlRaoParametersImport.importData`?

> Note that `importData` expects an input stream whereas `load` expects a `PlatformConfig`.

@pet-mit what's your take on this?

bqth29, 2025-02-03:
Besides, do we want to create a YAML exporter?