powsybl/powsybl-open-rao #1208: Gradient constraints

Godelaine, 2024-12-03:
### Describe the expected behavior

1) Data Management
Load multiple cracs, multiple networks, make a multi timestamp unit object : RaoInputTS
add gradient information for now in RaoInput : map< network element, double> 

2) Launch a multi timestamp sensitivity computation 
document difference % SensitivityComputation

3) use LinearProblemBuilderMultiTS
document difference % LinearProblemBuilder

4) GradientFiller
puissance(timeindex3)  - puissance(timeindex2) < gradient * (timeindex3 - timeindex2)

5) Multi TS Output
for now, list raoResults

For now, no topos
Tests :
* cnec with max -1000, injection on first node
* same, but new cnec in the middle with opposite (and impossible to meet) needs 

FOCUS ON CREATED OBJECTS:
1)RaoInputTS : Map<cracs, ts>, networks. For now, no optimized state etc (used for CastorOneStateOnly)
TODO : list other attributes in RaoInput and why we don't need them
2)

bqth29, 2024-12-05:
For power gradient constraints, I think that a simple record class can be used with the name `PowerGradientConstraint`. It should have 5 attributes:

- a string that contains the identifier of the network element (generator or load)
- a double value for the gradient value
- a variation direction (upward / downward) to whether the constraint applies to upward or downward set-point variations
- an optional timestamp to indicate from when the constraint applies (no timestamp means available all the time, until a possible end timestamp)
- an optional timestamp to indicate until when the constraint applies (no timestamp means available all the time, from a possible start timestamp)

```java
public record PowerGradientConstraint(String networkElementId, double powerGradient,
                                      RangeAction.VariationDirection variationDirection,
                                      Optional<OffsetDateTime> validFrom,
                                      Optional<OffsetDateTime> validTo) {
}
```

bqth29, 2024-12-05:
# Linear Equation Update

## Set-point value

The set-point and variation variables must now include a timestamp indexing alongside the range action and the state since the same remedial action can be used at different timestamps.

The set-point variation equation is the same as previously, linking the new set-point to the previous one using the variation variables:

$$A(r, s, t) = \pi(r, s, t) + \Delta^{+}(r, s, t) - \Delta^{-}(r, s, t)$$

where $\pi(r, s, t)$ is a function that returns the variable corresponding to the last time when the range action $r$ was used (time that occurs before the couple state $s$ - timestamp $t$).

> The equation is rigorously the same as for single-timestamp RAO, except for the timestamp-wise indexing.

The function $\pi$ is implemented in the utility class [`RaoUtil`](https://github.com/powsybl/powsybl-open-rao/blob/main/ra-optimisation/search-tree-rao/src/main/java/com/powsybl/openrao/searchtreerao/commons/RaoUtil.java#L180) for single-timestamp RAO and returns the variable corresponding to the last state at which the range action was available (or a double value corresponding to the initial set-point value in the network in case it was not available before).

For multi-timestamp RAO, the "last time the range action was used" is computed a bit differently. Indeed the time logic is now _two-dimensional_ as it relies on both the state **and** the timestamp (whereas only states were used before). Thus for a given state $s$ and a given timestamp $t$, the notion of "state just before $s$" is computed as follows:

- if state $s$ is not preventive, then the previous state is the same as previously (i.e. instant just before for the same timestamp and same contingency - or preventive state for the same timestamp)
- if state $s$ is preventive, then the previous state is the **preventive state of the timestamp just before $t$** (that we can denote by $t - 1$)
data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAArsAAACBCAYAAADAHXJgAAAY7klEQVR4Xu2dDZQeVX3G5w0hCDRGQ0sktIZqFdpjhWyoUrUCB7C0tYqfoHhMPIof9CDUnhZtCwlGOUGsB8FWqoAB/LatKLVFCyT1Ay3HTbB+tWotByEh4KEKGpOQZPs8ywz7suz77rzvO7Nz753fnPOc2d135s69v//973327p2ZTsYGAQhAAAIQgAAEIACBRAl0Em0XzYIABCAAAQhAAAIQgECG2aUTQAACEIAABCAAAQgkSwCzm2xoaRgEIAABCEAAAhCAAGaXPgABCEAAAhCAAAQgkCwBzG6yoaVhEIAABCAAAQhAAAKYXfoABCAAAQhAAAIQgECyBDC7yYaWhkEAAhCAAAQgAAEIYHbpAxCAAAQgAAEIQAACyRIIzewuE+nbZ6B9g372fOli6RXSQmmzdLa0qet4/3yH9GCyEaNhbSYwbH44dy6VDpKul16X50mbWdL29AgMmx8vFIoLpV+V/lE6Q9qTHh5aBIH2EgjN7O6nUByTh+OftP+y9B7pPulo6SrpLOlOaa20QDo8P/5p2t8ivUi6qb0hpeUJExgmP44Qj3ulm6VrJOfVX0iXJMyJprWTwDD5sVyofpznw3e0v1Z6sfTpdiKk1RBIk0BoZreb8lZ9c530pvyH79T+LdKTJH92pPQc6f3SjdLx+XEnao/ZTbO/0qopAmXz41M65SvSa6UvSh7QN0h/AkwIJEygbH58RAz+TLpc8nj4I8kzvZ9NmA1Ng0DrCMRkdp+RD9L7aH+r5FncK6XvS0ulX5c8E4zZbV03bmWDpw/m/fKjAPREffFDyf+m/VArqdHothAYND+eLTAfk26XPHHCMoa29BTa2QoCMZldB8Sm1usPnyudIu2SnpDvi/VamN1WdN3WN3L6YD5bfvg/Iv8q3SWdnOdM6yECIFkCg+aHJ0veIHmW1zO7/5IsGRoGgRYSiMnsvkDxsaG9LI/TCdp7+YKN75fyz27XHrPbwo7cwiZPH8z75cf94vOFPE9epb1v4mSDQMoEyuaHJ08Ok7zc5x7pW5L/Q/jGlOHQNgi0jUBMZtdrDN8nnZ//MvIaxJdJT5Z8wxozu23rve1u7/TBvF9+eJbqsdJKabe0TfpBu/HR+sQJlM2Po8TBT/T5qPR56cPSedJFifOheRBoFYGYzO48RWaddLq0WPJa3dVScdcsZrdVXbf1jZ0+mPfKD9+cZnPbvV2hb7xulw0CqRIomx8eP16Zjy2LtP+c5EfzbU8VDO2CQBsJhGx22xgP2gwBCEAAAhCAAAQgUCEBzG6FMCkKAhCAAAQgAAEIQCAsApjdsOJBbSAAAQhAAAIQgAAEKiSA2a0QJkVBAAIQgAAEIAABCIRFoE6ze6qa+nfSFukA6RfSKunrc4Dg8brGsyTfbOA3rPkOW9+owwaBUAiQH6FEgnqESID8CDEq1AkCkRKo2+z6db5n5Wy894sg/HzcYpuvL/wopKq3w1XgOyQ/mmx/yS+f4I04VVOmvFEIeDAnP0YhyLkpEyA/Uo4ubYPAHBOYS7P7FLXNzzH0m5z8EoibpCXSB6UPSPtK/y6dLW2UzpS+LfmlEW+W/JDv9ZIfMeb3l/vh+PdJX5Nuk54u+eH5fri+jztWeqvkGV7P7F44QJk6lA0CtRKYPpiTH7XipvDICJAfkQWM6kIgZAJzZXYPFIS10hHSH0qbpUukq6VvSqdJ35X8/M+vSgulfaSLc3npg02vj71cWiWdJPmZuy7LL5u4Mj//M9p/TypmdotlDM8coMyQY0bd0iDQPZiTH2nElFZUR4D8qI4lJUGg9QTqNrvFml2D/l/pLZLf3GSDelxO328/uyH/+te0/4b0rty4Hq+9325TvBL4Ln3ttb+PkZ4qebmCy/oD6W7pXMnl2RxPN7v+3Ga6TJmt7xgAqJ1A95pE8qN23FwgMgLkR2QBo7oQCJlA3Wa3e01iNwcb1KMlr9m9RVqRf+ilDJ7R3SF5aYJfCfw26aX5cSdr76UKrrdnw34muaxjpJ1SP7PrG9TKlhlyzKhbGgSm/5uW/EgjrrSiGgLkRzUcKQUCEMhNY10gyvyy8k1jXrbg9bie0b1Wul76pOQ1tp7RvUz6hOTZ3jskL1mwCbZB9rreXmbX579E6n4aQ9ky62JCuRAoCJAf9AUI9CZAftA7IACBygg0PbNrs3uk5BvU/HiyW6UzpL2S19hukA6WPIPrx4ldIy3Nv3+59tt6mF0bZptoG2TP+haPHitbZmWAKQgCPQiUHczJD7pQGwmQH22MOm2GQE0E6jS7NVWZYiEAAQhAAAIQgAAEIFCOAGa3HCeOggAEIAABCEAAAhCIkABmN8KgUWUIQAACEIAABCAAgXIEMLvlOHEUBCAAAQhAAAIQgECEBDC7EQaNKkMAAhCAAAQgAAEIlCOA2S3HiaMgAAEIQAACEIAABCIkgNmNMGhUGQL9CEyMjd2cdTp+UyAbBCDwaAIbO+Pj5Ac9AwItIoDZbVGwaWo7CEysWDGhwZzcbke4aeWABMiPAYFxOAQSIMCAmEAQaQIEugkwmNMfINCbAPlB74BA+whgdtsXc1qcOAEG88QDTPNGIkB+jISPkyEQJQHMbpRho9IQYOaKPgCBYQhgdoehxjkQiJsAZjfu+FF7CDyKAIM5nQIC/DFIH4AABKYIYHbpDRBIjABmN7GA0pxKCZAfleKkMAhEQQCzG0WYqCQEyhNgMC/PiiPbR4D8aF/MaTEEMLv0AQgkRoDBPLGA0pxKCZAfleKkMAhEQQCzG0WYqCQEyhNgMC/PiiPbR4D8aF/MaTEEMLv0AQgkRoDBPLGA0pxKCZAfleKkMAhEQQCzG0WYqCQEyhNgMC/PiiPbR4D8aF/MaTEEMLv0AQgkRoDBPLGA0pxKCZAfleKkMAhEQQCzG0WYqCQEyhNgMC/PiiPbR4D8aF/MaTEEMLv0AQgkRoDBPLGA0pxKCZAfleKkMAhEQQCzG0WYqCQEyhNgMC/PiiPbR4D8aF/MaTEEMLv0AQgkRoDBPLGA0pxKCZAfleKkMAhEQQCzG0WYqCQEyhNgMC/PiiPbR4D8aF/MaTEEMLv0AQgkRoDBPLGA0pxKCZAfleKkMAhEQQCzG0WYqCQEyhNgMC/PiiPbR4D8aF/MaTEEYjW7qxW6C1oevmeq/e+SDpJ2Sn8vfaAPk8frs2dJnwuA20dVh2dIvzFLXeqo85t1zddJ86W7pddLPxiBSXcd369yPix9ZYTyRj6VwXwSIfkxXE8iP4bjFttZ5MdwEUs+P4QlSX8Vq9mdUEBirftwKfbIsw7Xt9dLL5K+LT1O+rh0nXR5jwv4nHdIL6uiAiOUsZ/O/aF0n/RK6Zt9yqq6zm77a6UXS9ulP5beKj27RHtsjnfPcFx3HffX57ukPSXKq+0QzG5GfgzXu8iP4bjFdhb5MVzEWpEfQpOkv4rVMCYZjAHy70ode5PkGdJiW6ovviw9KTeQv51/sFb7r0unSsfm5u5W7a+SFkg2nS+Xfir5r9aV0r3SIZJngm3g1kvLpB9 [... cut ...] ZnckfJwMgfAIYHbDiwk1ggAEIACB5ghgdptjz5UhUAsBzG4tWCkUAhCAAAQiJYDZjTRwVBsCvQhgdukbEIAABCAAgSkCcZrdsbGbs07neAIJAQjMSGBjZ3yc/KBzQAACEIAABEQgSrNL5CAAAQhAAAIQgAAEIFCGAGa3DCWOgQAEIAABCEAAAhCIkgBmN8qwUWkIQAACEIAABCAAgTIEMLtlKHEMBCAAAQhAAAIQgECUBDC7UYaNSkMAAhCAAAQgAAEIlCGA2S1DiWMgAAEIQAACEIAABKIkgNmNMmxUGgIQgAAEIAABCECgDAHMbhlKHAMBCEAAAhCAAAQgECUBzG6UYaPSEIAABCAAAQhAAAJlCGB2y1DiGAhAAAIQgAAEIACBKAlgdqMMG5WGAAQgAAEIQAACEChDALNbhhLHQAACEIAABCAAAQhESQCzG2XYqDQEIAABCEAAAhCAQBkC/w/QCuonVySwbwAAAABJRU5ErkJggg==

bqth29, 2024-12-05:
# Handling different data for each timestamp

Some classes must be used for several timestamps (CRAC, network, RAO parameters, RAO result, ...) so it would make sense using a general interface for handling temporal data.

For this purpose, we can take inspiration form PowSyBl's [`ZonalData`](https://github.com/powsybl/powsybl-entsoe/blob/main/glsk/glsk-commons/src/main/java/com/powsybl/glsk/commons/ZonalData.java) and create an equivalent system for time data that can be called `TemporalData`.

```java
public interface TemporalData<T> {
    Map<OffsetDateTime, T> getDataPerTimestamp();

    default Optional<T> getData(OffsetDateTime timestamp) {
        return Optional.ofNullable(getDataPerTimestamp().get(timestamp));
    }

    void add(OffsetDateTime timestamp, T data);
}
```

That way, there will be a common handling of temporal data, no matter which class is being processed.

bqth29, 2024-12-05:
# Outputs

For the first implementation, the easiest is to simply return a `TemporalData<RaoResult>`.

bqth29, 2024-12-05:
# Input data

closed 2025-02-11 by Godelaine