powsybl/powsybl-open-rao #1209: Multi TS skeleton

Godelaine, 2024-12-03:
### Describe the expected behavior

1st iteration: only preventive 
1)N (number of timestamps) CASTOR with only preventive network actions
2) Output : N Rao Results
3) 1 Computation with N RaoInputs with only linear actions.

bqth29, 2024-12-09:
For topological optimization, the strategy we wanted to use was to optimize all 24 timestamps in parallel to save as must computation time as possible and then to merge everything in a single linear problem to optimize range actions.

This can be done without too much work. However, even if it will may not be part of the first POC, we have to keep in mind that there will possibly be constraints in terms of remedial action changes between two consecutive timestamps.

bqth29, 2024-12-09:
How to consider that a RA was activated between 2 consecutive TS?

PR #1258 (2025-02-20): Inter-Temporal Skeleton (MARMOT)
**Please check if the PR fulfills these requirements**

- [ ] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**
Fixes #1209 

**What kind of change does this PR introduce?**

**What is the current behavior?**

**What is the new behavior (if this is a feature change)?**

**Does this PR introduce a breaking change or deprecate an API?**
- [ ] Yes
- [ ] No

**If yes, please ch [... cut]
closed 2025-02-11 by Godelaine