powsybl/powsybl-open-rao #1216: Create `intertemporal-rao` module

bqth29, 2024-12-06:
### Describe the expected behavior

As a preliminary step for the upcoming developments on intertemporality, the `intertemporal-rao` module must be created. The paths all Java files shall all start with `com.powsybl.openrao.intertemporal`.

Godelaine, 2024-12-06:
why not disseminate developments in existing modules ?

bqth29, 2024-12-06:
> why not disseminate developments in existing modules ?

Even if we seek to reuse as many existing objects and classes as possible, I thought that we would sort of make an independent implementation. However I agree with you that this could also be spread in various existing modules.

I believe it requires more discussions and that we should align on the same visions.

Godelaine, 2024-12-06:
How about we start by including developments in existing modules, and we'll reevaluate the situation when we're further in the developments ?

closed 2024-12-06 by bqth29