powsybl/powsybl-open-rao #1218: Power Gradient Constraints

bqth29, 2024-12-06:
### Describe the expected behavior

As introduced in #1208 (in [this comment](https://github.com/powsybl/powsybl-open-rao/issues/1208#issuecomment-2519958938)), the intertemporal RAO will require Power Gradient Constraints (PGCs).

Implement the `PowerGradientConstraint` class in `intertemporal-rao`.

### Extra Information

After some discussions, we concluded that there this no need, at first, to have a start and end validity dates for the constraints (though they may be required for other types of intertemporal constraints). Thus, the `PowerGradientConstraint` shall only have 3 attributes: the network element, the gradient value and the variation direction.

```java
public record PowerGradientConstraint(String networkElementId, double powerGradient, RangeAction.VariationDirection variationDirection) {
}
```

Besides, it could be worth to isolate the `VariationDirection` enum out of the `RangeAction` i,nterface in a standalone file in `crac-api`.

> Wait for #1186 to be merged before to address this issue.

Godelaine, 2024-12-06:
Is the reason why you want to separate VariationDirection to reuse it for other constraints ?

bqth29, 2024-12-06:
> Is the reason why you want to separate VariationDirection to reuse it for other constraints ?

No, it is because currently, the enum is part of the RangeAction interface. As it can also be used for gradient constraints I thought it would be cleaner to get it out as it will no longer be a specific RangeAction's field.

PR #1229 (2024-12-19): Power Gradient Constraint
**Please check if the PR fulfills these requirements**

- [ ] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)

**Does this PR already have an issue describing the problem?**
Fixes #1218

**What kind of change does this PR introduce?**

**What is the current behavior?**

**What is the new behavior (if this is a feature change)?**

**Does this PR introduce a breaking change or deprecate an API?**
- [ ] Yes
- [ ] No

**If yes, please check if t [... cut]
closed 2024-12-19 by Godelaine