powsybl/powsybl-open-rao #1291: Ignore duplicated CNECs in RAO status

terriervik, 2025-02-27:
### Describe the current behavior

CNECs at instant auto are duplicated so that they can be included in the preventive optimisation perimeter, which is especially important for the CNECs having no linked ARA (because there is no RA available to solve them in any perimeter).
Currently, the flows of these duplicated CNECs are not reported in the CNE file, but they are taken into account in the global status of the RAO (secure or unsecure) and written in the RaoResult file.

Thus, if a duplicated CNEC is unsecure, the users see that the RAO is unsecure, even though there is possibly no unsecure element in the CNE file...

### Describe the expected behavior

The status of the duplicated CNECs should not be used anymore in the definition of the status of the RAO.
Besides, the logs should be checked and updated if they are not clear enough for the users to understand the duplication (to be discussed together).

Godelaine, 2025-03-03:
cas de figure : quand le most limiting element est une cnec dupliquée
il faut :
CastorFullOpt
* checker que le most limiting elment contient "outage duplicate"
* excludeCnecs(cnecs fictives) comme contingencies à la fin de CastorFullOpt
* rappeler getCost 
* vérifier que le statut est ok (on se basera sur les cost results et non sur les flow results)

LOGS
rajouter un log final cost is reevaluated : blabla

RAORESULT.json
tous les costResults sont automatiquement modifiés via le excludeCnecs 

CNE
* vérifier qu'on ne reporte pas les cnecs fictives
* recalculer le cout sans les cnecs fictivies via la reevaluation de la fonction objectif
* recalculer le statut

Godelaine, 2025-03-13:
+ delete duplicate cnecs from raoResult

PR #1303 (2025-03-20): Ignore duplicated CNECs in RAO status
**Please check if the PR fulfills these requirements**

- [ ] The commit message follows our guidelines
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] A PR or issue has been opened in all impacted repositories (if any)

**Does this PR already have an issue describing the problem?**

Fixes #1291 

**What kind of change does this PR introduce?**

**What is the current behavior?**

**What is the new behavior (if this is a feature change)?**

**Does this PR introduce  [... cut]
closed 2025-03-20 by bqth29